stack-of-tasks/pinocchio #2635: Support for CasADi symbolic expressions in CRBA algorithm

LockedFlysher, 2025-04-03:
## Description

### Problem Description
I'm trying to use Pinocchio with CasADi symbolic expressions to obtain the symbolic mass matrix through the CRBA algorithm. However, I've encountered an issue where the `crba()` function doesn't seem to accept CasADi symbolic expressions as input for the `q` parameter.

Here's a minimal code example that demonstrates the issue:

```cpp
bool ManipulatorModel::buildModel(const std::string &urdfPath) {
    FixedBaseRobot::parseURDF(urdfPath);
    optVars.q = ::casadi::SX::sym("q", modelPtr_->nq);
    optVars.dq = ::casadi::SX::sym("dq", modelPtr_->nv);
    optVars.F = ::casadi::SX::sym("F", modelPtr_->nq);
    Eigen::Matrix<::casadi::SX,-1,1> q_copy;
    pinocchio::casadi::copy(optVars.q, q_copy);
    
    // This line fails because crba doesn't accept CasADi symbolic expressions
    pinocchio::crba(*modelPtr_, *dataPtr_, optVars.q);
    return true;
}
```

When trying to call `pinocchio::crba()` with a CasADi symbolic expression as the `q` parameter, the compilation fails due to type incompatibility.

### Expected Behavior
I would expect the CRBA algorithm to work with CasADi symbolic expressions, similar to how other functions in Pinocchio support CasADi for automatic differentiation and symbolic computation.

### Current Behavior
The CRBA function only accepts numerical values for the `q` parameter, making it impossible to obtain a symbolic expression for the mass matrix directly.

### Question
Is there a way to use the CRBA algorithm with CasADi symbolic expressions in Pinocchio? If not, could this feature be added in future releases?

### Additional Context
I'm working on an optimal control problem where having the symbolic expression of the mass matrix would be extremely valuable for gradient computation and optimization. Currently, I have to resort to workarounds that are less efficient and more complex than having direct symbolic support in the CRBA algorithm.

### Environment
- Pinocchio version: [version]
- CasADi version: [version]
- Operating system: [OS]
- Compiler: [compiler and version]

Thank you for considering this feature request. I would be happy to provide additional information or assist with implementation if needed.

jcarpent, 2025-04-03:
```cop
bool ManipulatorModel::buildModel(const std::string &urdfPath) {
    FixedBaseRobot::parseURDF(urdfPath);
    optVars.q = ::casadi::SX::sym("q", modelPtr_->nq);
    optVars.dq = ::casadi::SX::sym("dq", modelPtr_->nv);
    optVars.F = ::casadi::SX::sym("F", modelPtr_->nq);
    Eigen::Matrix<::casadi::SX,-1,1> q_copy;
    pinocchio::casadi::copy(optVars.q, q_copy);
    
    // This line fails because crba doesn't accept CasADi symbolic expressions
    pinocchio::crba(*modelPtr_, *dataPtr_, q_copy);
    return true;
}
```
should work.

LockedFlysher, 2025-04-04:
Thank you for looking into this issue. I'm still encountering the template error when trying to obtain the symbolic expression of the mass matrix M(q) for use with CasADi's `Function` to implement a NMPC solver.

## Current Type Definitions
```cpp
std::shared_ptr<ModelTpl<::casadi::SX>> modelPtr_;
std::shared_ptr<ModelTpl<::casadi::SX>::Data> dataPtr_;
```

Which corresponds to:
```cpp
std::shared_ptr<ModelTpl<casadi::Matrix<casadi::SXElem>>> modelPtr_
std::shared_ptr<DataTpl<casadi::Matrix<casadi::SXElem>>> dataPtr_
```

And I'm working with:
```cpp
Eigen::Matrix<casadi::Matrix<casadi::SXElem>, -1, 1> q_copy
```

## My Question

I'm wondering if what I'm trying to achieve - obtaining a symbolic expression of the mass matrix M(q) for use with CasADi - is already implemented somewhere in the Pinocchio repository? My ultimate goal is to build a CasADi `Function` that I can use `q`  to easily get `M(q)`, and then it's much eazier to build an MPC problem, I can show you some of my successful example .
```cpp
// centroidal dynamics
    // 1. COM function
    ::casadi::SX comPosSX;
    pinocchio::casadi::copy(dataPtr_->com[0],comPosSX);
    comPosKinematicFunction_ = ::casadi::Function(
            "comPosKinematicConstraintFunction",
            {centroidalAttributes_.q},
            {comPosSX});
// 2. Jacobian function
Eigen::Matrix<::casadi::SX,-1,1> q_copy;
    pinocchio::casadi::copy(centroidalAttributes_.q,q_copy);
    std::vector<pinocchio::FrameIndex> contactFrameID;
    pinocchio::computeJointJacobians(*modelPtr_,
                                     *dataPtr_,
                                     q_copy);

    for (const auto &contactFrameName: contactFrameNames_) {
        pinocchio::FrameIndex frameID = modelPtr_->getFrameId(contactFrameName);
        contactFrameID.push_back(frameID);
        auto jacobianEigenSX = pinocchio::getFrameJacobian(*modelPtr_,
                                                           *dataPtr_,
                                                           frameID,
                                                           LOCAL_WORLD_ALIGNED);
        ::casadi::SX jacobianSX(6, modelPtr_->nv);
        ::pinocchio::casadi::copy(jacobianEigenSX, jacobianSX);
        jacobianFunctions_.push_back(::casadi::Function("JacobianForContactPoint",
                                                        {centroidalAttributes_.q},
                                                        {jacobianSX}));
    }
```

If so, could you kindly point me to the relevant examples or documentation? If not, I would greatly appreciate any guidance on how to properly handle these template types to solve the error shown in the screenshots.

Thank you very much for your assistance.
[image]
[image]

jcarpent, 2025-04-04:
@LockedFlysher You have a full working example in the unit tests of Pinocchio. Please have a look to:
https://github.com/stack-of-tasks/pinocchio/blob/733891651b9262e9c4e07ae6b6312b9a5a722670/unittest/casadi/algorithms.cpp#L264-L355

jcarpent, 2025-04-04:
@LockedFlysher For your convenience, I've added an example of CRBA+CasADi in https://github.com/stack-of-tasks/pinocchio/pull/2637

jcarpent, 2025-04-07:
I will close the issue as there is no issue with Pinocchio.

closed 2025-04-07 by jcarpent

LockedFlysher, 2025-04-12:
Thanks!!!!!!!!!!!!!!!!!! This really helped me, I came back from holiday, sorry for this late response to you. 
I have finished my project, used exported casadi function building dynamics for manipulator, it run on GPU, great