stack-of-tasks/pinocchio #2653: module 'pinocchio' has no attribute 'buildModelFromMjcf'

Wan020607, 2025-04-17:
## Bug description
My Ubuntu version is 20.04. I created a Python 3.9 virtual environment and installed Pinocchio in the environment. I want to use MJCF to describe closed-loop robot. When I run my code, I get the following error:
```
    model, constraint_models = pin.buildModelFromMjcf(mjcf_model_path, root_link_name)
AttributeError: module 'pinocchio' has no attribute 'buildModelFromMjcf'
```

## Expected behavior

I would like to calculate the generalized mass matrix and gravity part of the dynamics equation of a parallel robot using pinocchio.

## Reproduction steps

I tried three ways to install pinocchio

**Method 1**: python -m pip install pin
result:
```
    model, constraint_models = pin.buildModelFromMjcf(mjcf_model_path, root_link_name)
AttributeError: module 'pinocchio' has no attribute 'buildModelFromMjcf'
```

**Method 2**: conda install pinocchio -c conda-forge
result: the result of Method 2 is same as Method1.

**Method3**: Install from source
Referring to the official tutorial, I wrote a script to install pinocchio in a virtual environment shown in Code.
result: the result of Method 2 is same as Method1. 
And I run the example code of pinocchio: simulation-closed-kinematic-chains.py, the error is shown as follows:

```
Traceback (most recent call last):
  File "/home/wan/closed_loop_test/src/script/simulation_closed.py", line 47, in <module>
    geom_obj0 = pin.GeometryObject(
Boost.Python.ArgumentError: Python argument types in
    GeometryObject.__init__(GeometryObject, str, int, SE3, Capsule)
did not match C++ signature:
    __init__(_object* self, pinocchio::GeometryObject otherGeometryObject)
    __init__(_object* self, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > name, unsigned long parent_joint, std::shared_ptr<pinocchio::fcl::FakeCollisionGeometry> collision_geometry, pinocchio::SE3Tpl<double, 0> placement)
    __init__(_object* self, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > name, unsigned long parent_joint, std::shared_ptr<pinocchio::fcl::FakeCollisionGeometry> collision_geometry, pinocchio::SE3Tpl<double, 0> placement, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > mesh_path)
    __init__(_object* self, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > name, unsigned long parent_joint, std::shared_ptr<pinocchio::fcl::FakeCollisionGeometry> collision_geometry, pinocchio::SE3Tpl<double, 0> placement, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > mesh_path, Eigen::Matrix<double, 3, 1, 0, 3, 1> mesh_scale)
    __init__(_object* self, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > name, unsigned long parent_joint, std::shared_ptr<pinocchio::fcl::FakeCollisionGeometry> collision_geometry, pinocchio::SE3Tpl<double, 0> placement, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > mesh_path, Eigen::Matrix<double, 3, 1, 0, 3, 1> mesh_scale, bool override_material)
    __init__(_object* self, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > name, unsigned long parent_joint, std::shared_ptr<pinocchio::fcl::FakeCollisionGeometry> collision_geometry, pinocchio::SE3Tpl<double, 0> placement, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > mesh_path, Eigen::Matrix<double, 3, 1, 0, 3, 1> mesh_scale, bool override_material, Eigen::Matrix<double, 4, 1, 0, 4, 1> mesh_color)
    __init__(_object* self, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > name, unsigned long parent_joint, std::shared_ptr<pinocchio::fcl::FakeCollisionGeometry> collision_geometry, pinocchio::SE3Tpl<double, 0> placement, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > mesh_path, Eigen::Matrix<double, 3, 1, 0, 3, 1> mesh_scale, bool override_material, Eigen::Matrix<double, 4, 1, 0, 4, 1> mesh_color, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > mesh_texture_path)
    __init__(_object* self, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > name, unsigned long parent_joint, std::shared_ptr<pinocchio::fcl::FakeCollisionGeometry> collision_geometry, pinocchio::SE3Tpl<double, 0> placement, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > mesh_path, Eigen::Matrix<double, 3, 1, 0, 3, 1> mesh_scale, bool override_material, Eigen::Matrix<double, 4, 1, 0, 4, 1> mesh_color, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > mesh_texture_path, boost::variant<pinocchio::GeometryNoMaterial, pinocchio::GeometryPhongMaterial> mesh_material)
    __init__(_object*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > self, unsigned long name, unsigned long parent_frame, std::shared_ptr<pinocchio::fcl::FakeCollisionGeometry> parent_joint, pinocchio::SE3Tpl<double, 0> collision_geometry)
    __init__(_object*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > self, unsigned long name, unsigned long parent_frame, std::shared_ptr<pinocchio::fcl::FakeCollisionGeometry> parent_joint, pinocchio::SE3Tpl<double, 0> collision_geometry, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > placement)
    __init__(_object*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > self, unsigned long name, unsigned long parent_frame, std::shared_ptr<pinocchio::fcl::FakeCollisionGeometry> parent_joint, pinocchio::SE3Tpl<double, 0> collision_geometry, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > placement, Eigen::Matrix<double, 3, 1, 0, 3, 1> mesh_path)
    __init__(_object*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > self, unsigned long name, unsigned long parent_frame, std::shared_ptr<pinocchio::fcl::FakeCollisionGeometry> parent_joint, pinocchio::SE3Tpl<double, 0> collision_geometry, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > placement, Eigen::Matrix<double, 3, 1, 0, 3, 1> mesh_path, bool mesh_scale)
    __init__(_object*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > self, unsigned long name, unsigned long parent_frame, std::shared_ptr<pinocchio::fcl::FakeCollisionGeometry> parent_joint, pinocchio::SE3Tpl<double, 0> collision_geometry, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > placement, Eigen::Matrix<double, 3, 1, 0, 3, 1> mesh_path, bool mesh_scale, Eigen::Matrix<double, 4, 1, 0, 4, 1> override_material)
    __init__(_object*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > self, unsigned long name, unsigned long parent_frame, std::shared_ptr<pinocchio::fcl::FakeCollisionGeometry> parent_joint, pinocchio::SE3Tpl<double, 0> collision_geometry, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > placement, Eigen::Matrix<double, 3, 1, 0, 3, 1> mesh_path, bool mesh_scale, Eigen::Matrix<double, 4, 1, 0, 4, 1> override_material, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > mesh_color)
    __init__(_object*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > self, unsigned long name, unsigned long parent_frame, std::shared_ptr<pinocchio::fcl::FakeCollisionGeometry> parent_joint, pinocchio::SE3Tpl<double, 0> collision_geometry, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > placement, Eigen::Matrix<double, 3, 1, 0, 3, 1> mesh_path, bool mesh_scale, Eigen::Matrix<double, 4, 1, 0, 4, 1> override_material, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > mesh_color, boost::variant<pinocchio::GeometryNoMaterial, pinocchio::GeometryPhongMaterial> mesh_texture_pathmesh_material)
    __init__(_object* self, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > name, unsigned long parent_joint, pinocchio::SE3Tpl<double, 0> placement, std::shared_ptr<pinocchio::fcl::FakeCollisionGeometry> collision_geometry)
    __init__(_object* self, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > name, unsigned long parent_joint, pinocchio::SE3Tpl<double, 0> placement, std::shared_ptr<pinocchio::fcl::FakeCollisionGeom [... cut ...]
## Additional context

Add any other context about the problem here.

### The Reason why I have to install pinocchio in virtual environment

I compiled ocs2 in my base environment, where the pinocchio environment dependency used by ocs2 is an old version, so if I compile a new version of pinocchio in my base environment, it may conflict with ocs2's pinocchio, resulting in errors in the compilation of ocs2-based C++ projects.

I will appreciate it if you can read my bug and present your views and suggestions!!!!

MegMll, 2025-04-17:
Hi, the function you're trying to use does not exist, i believe. Could you try this instead, with MJCF written in full caps ? 

```
import pinocchio as pin
from pinocchio.visualize import MeshcatVisualizer
import numpy as np
import os
import time

model_path = "../urdf"
mjcf_filename = "hexapodrobot.xml"
mjcf_model_path = os.path.join(model_path, mjcf_filename)
root_link_name = "LF1_base_link"

model, constraint_models = pin.buildModelFromMJCF(mjcf_model_path, root_link_name)
```

Wan020607, 2025-04-18:
/

Wan020607, 2025-04-18:
thanks for your help, it works when I use buildModelFromMJCF function. And I want to ask if there is an API for Python

MegMll, 2025-04-18:
i'm not sure i understand what you mean by python API, because you are using it, i think

jcarpent, 2025-04-19:
@Wan020607 `pin.buildModelFromMJCF` is part of the Python API.

closed 2025-04-19 by jcarpent