stack-of-tasks/pinocchio #2660: Segmentation fault when using pinocchio::mjcf::buildModel

midiexiangxxx, 2025-04-22:
## Bug description

Hello! I'm working with MuJoCo 3.3.1 and C++. I have my mujoco build from source and pinocchio install with apt. 
The code that went wrong was just like: 
```
  const char* filename = nullptr;
  if (argc >  1) {
    filename = argv[1];
    // scene_path = argv[2];
  }
  const char* pin_model = argv[2];
  pin::Model model;
  pin::mjcf::buildModel(pin_model, model);
```
and running this code I encountered a segmentation fault.

I've searched many web pages and still have no clue.

I can provide my back trace with gdb:

```
Program received signal SIGSEGV, Segmentation fault.
__GI___libc_free (mem=0x81) at malloc.c:3102
3102    malloc.c: No such file or directory.
(gdb) br
Breakpoint 1 at 0x7ffff69cf6f0: file malloc.c, line 3102.
(gdb) bt
#0  __GI___libc_free (mem=0x81) at malloc.c:3102
#1  0x0000555555565269 in Eigen::internal::handmade_aligned_free (ptr=0x555555da98a0) at /usr/include/eigen3/Eigen/src/Core/util/Memory.h:98
#2  0x00005555555652d1 in Eigen::internal::aligned_free (ptr=0x555555da98a0) at /usr/include/eigen3/Eigen/src/Core/util/Memory.h:179
#3  0x000055555556be74 in Eigen::internal::conditional_aligned_free<true> (ptr=0x555555da98a0) at /usr/include/eigen3/Eigen/src/Core/util/Memory.h:230
#4  0x000055555556ef82 in Eigen::internal::conditional_aligned_delete_auto<double, true> (ptr=0x555555da98a0, size=14)
    at /usr/include/eigen3/Eigen/src/Core/util/Memory.h:416
#5  0x000055555556b8f9 in Eigen::DenseStorage<double, -1, -1, 1, 0>::~DenseStorage (this=0x7fffffffb9f0, __in_chrg=<optimized out>)
    at /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:542
#6  0x0000555555565584 in Eigen::PlainObjectBase<Eigen::Matrix<double, -1, 1, 0, -1, 1> >::~PlainObjectBase (this=0x7fffffffb9f0, __in_chrg=<optimized out>)
    at /usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:98
#7  0x00005555555655a4 in Eigen::Matrix<double, -1, 1, 0, -1, 1>::~Matrix (this=0x7fffffffb9f0, __in_chrg=<optimized out>)
    at /usr/include/eigen3/Eigen/src/Core/Matrix.h:178
#8  0x0000555555577472 in pinocchio::mjcf::details::MjcfGraph::~MjcfGraph (this=0x7fffffffb7c0, __in_chrg=<optimized out>)
    at /opt/openrobots/include/pinocchio/parsers/mjcf/mjcf-graph.hpp:381
#9  0x00005555555775e9 in pinocchio::mjcf::buildModelFromXML<double, 0, pinocchio::JointCollectionDefaultTpl> (xmlStream="assets/ur10e_2f85.xml", model=..., 
    contact_models=std::vector of length 2, capacity 2 = {...}, verbose=false) at /opt/openrobots/include/pinocchio/parsers/mjcf/model.hxx:57
#10 0x00005555555730be in pinocchio::mjcf::buildModelFromXML<double, 0, pinocchio::JointCollectionDefaultTpl> (xmlStream="assets/ur10e_2f85.xml", model=..., verbose=false)
--Type <RET> for more, q to quit, c to continue without paging--
    at /opt/openrobots/include/pinocchio/parsers/mjcf/model.hxx:33
#11 0x000055555556e65c in pinocchio::mjcf::buildModel<double, 0, pinocchio::JointCollectionDefaultTpl> (filename="assets/ur10e_2f85.xml", model=..., verbose=false)
    at /opt/openrobots/include/pinocchio/parsers/mjcf/model.hxx:23
#12 0x000055555556b307 in main (argc=3, argv=0x7fffffffd3c8) at /home/run/Projects/VLAssembly/main.cc:553
```

OS version ubuntu 20.04.

This bug can be reproduced by simply adding a pinocchio::mjcf::buildModel to the main function of mujoco's simulate project. I've searched many web pages and still have no clue. Some suggestions are to remove the vectorization in eigen, but I don't have any “-march=native” flags in my cmakelists.

midiexiangxxx, 2025-04-22:
Thanks for replying!
My excutable was called simulate and 
`ldd simulate` gives:
```
        linux-vdso.so.1 (0x00007ffc86961000)
	libompl.so.15 => /lib/x86_64-linux-gnu/libompl.so.15 (0x00007f907f022000)
	libmujoco.so.3.3.1 => /home/run/mujoco_install/lib/libmujoco.so.3.3.1 (0x00007f907ec76000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f907ec70000)
	libX11.so.6 => /lib/x86_64-linux-gnu/libX11.so.6 (0x00007f907eb33000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f907eb10000)
	libpinocchio_parsers.so.3.4.0 => /opt/openrobots/lib/libpinocchio_parsers.so.3.4.0 (0x00007f907ea8c000)
	libpinocchio_default.so.3.4.0 => /opt/openrobots/lib/libpinocchio_default.so.3.4.0 (0x00007f907e5d5000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f907e367000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f907e218000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f907e1f3000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f907e001000)
	libode.so.8 => /lib/x86_64-linux-gnu/libode.so.8 (0x00007f907dedd000)
	libboost_serialization.so.1.67.0 => /lib/x86_64-linux-gnu/libboost_serialization.so.1.67.0 (0x00007f907de9a000)
	libboost_filesystem.so.1.67.0 => /lib/x86_64-linux-gnu/libboost_filesystem.so.1.67.0 (0x00007f907de7d000)
	libboost_system.so.1.67.0 => /lib/x86_64-linux-gnu/libboost_system.so.1.67.0 (0x00007f907de76000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f907f7a4000)
	libxcb.so.1 => /lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f907de4c000)
	libboost_filesystem.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_filesystem.so.1.71.0 (0x00007f907de2e000)
	liburdfdom_model.so.1.0 => /lib/x86_64-linux-gnu/liburdfdom_model.so.1.0 (0x00007f907de07000)
	libcoal.so.3.0.1 => /opt/openrobots/lib/libcoal.so.3.0.1 (0x00007f907d631000)
	libccd.so.2 => /lib/x86_64-linux-gnu/libccd.so.2 (0x00007f907d625000)
	libXau.so.6 => /lib/x86_64-linux-gnu/libXau.so.6 (0x00007f907d61f000)
	libXdmcp.so.6 => /lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f907d617000)
	libtinyxml.so.2.6.2 => /lib/x86_64-linux-gnu/libtinyxml.so.2.6.2 (0x00007f907d5fd000)
	libconsole_bridge.so.0.4 => /lib/x86_64-linux-gnu/libconsole_bridge.so.0.4 (0x00007f907d5f7000)
	libassimp.so.5 => /lib/x86_64-linux-gnu/libassimp.so.5 (0x00007f907c9aa000)
	libboost_serialization.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_serialization.so.1.71.0 (0x00007f907c966000)
	liboctomap.so.1.9 => /lib/x86_64-linux-gnu/liboctomap.so.1.9 (0x00007f907c918000)
	liboctomath.so.1.9 => /lib/x86_64-linux-gnu/liboctomath.so.1.9 (0x00007f907c910000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f907c8f4000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f907c8d8000)
	libminizip.so.1 => /lib/x86_64-linux-gnu/libminizip.so.1 (0x00007f907c6cd000)
```

`ldd libmujoco.so.3.3.1` gives:
```
	linux-vdso.so.1 (0x00007ffee43f7000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f55cfff6000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f55cfea7000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f55cfc39000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f55cfc14000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f55cfa22000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f55d03f6000)
```

nim65s, 2025-04-22:
Ok, so I'm not sure what you put in argv, but this is working for me:
```Dockerfile

# Setup Robotpkg apt binary repository
ADD http://robotpkg.openrobots.org/packages/debian/robotpkg.key /
RUN --mount=type=cache,sharing=locked,target=/var/cache/apt --mount=type=cache,sharing=locked,target=/var/lib/apt \
    apt-get update -qqy && DEBIAN_FRONTEND=noninteractive apt-get install -qqy gnupg2 \
 && apt-key add /robotpkg.key \
 && echo "deb [arch=amd64] http://robotpkg.openrobots.org/packages/debian/pub focal robotpkg" \
 >> /etc/apt/sources.list.d/robotpkg.list
ENV CMAKE_PREFIX_PATH=/opt/openrobots

# Install pinocchio
RUN --mount=type=cache,sharing=locked,target=/var/cache/apt --mount=type=cache,sharing=locked,target=/var/lib/apt \
    apt-get update -qqy && DEBIAN_FRONTEND=noninteractive apt-get install -qqy \
    cmake \
    robotpkg-py3\*-pinocchio

# Setup sources
WORKDIR /src
ADD https://raw.githubusercontent.com/stack-of-tasks/pinocchio-minimal/refs/heads/main/CMakeLists.txt .
ADD https://raw.githubusercontent.com/stack-of-tasks/pinocchio/refs/heads/master/unittest/models/test_mjcf.xml .
RUN <<EOF cat >> main.cpp
#include <pinocchio/parsers/mjcf.hpp>
int main(int argc, char *argv[]) {
    const char* filename = argv[1];
    pinocchio::Model model;
    pinocchio::mjcf::buildModel(filename, model);
    return 0;
}
EOF

# Compile the project
RUN cmake -B build
RUN cmake --build build

# Test it
RUN ./build/main test_mjcf.xml

midiexiangxxx, 2025-04-23:
Thanks. I finally found there is a `set(MUJOCO_SIMULATE_COMPILE_OPTIONS "${AVX_COMPILE_OPTIONS}" "${EXTRA_COMPILE_OPTIONS}")` line in the CMakeLists.txt, and simply comment it solves the problem. 
I'll close this issue since it's not a pinocchio problem.

closed 2025-04-23 by midiexiangxxx