stack-of-tasks/pinocchio #2673: No module named 'pinocchio.pinocchio_pywrap_default' when installiing using pip

kczttm, 2025-05-05:
## Bug description
```
(robosuite) ck@ks-X1E-Gen-3:~/workspaces/robosuite$ uv pip install pin
Resolved 13 packages in 6ms
Installed 13 packages in 165ms
 + cmeel==0.57.3
 + cmeel-assimp==5.4.3.1
 + cmeel-boost==1.87.0.1
 + cmeel-console-bridge==1.0.2.3
 + cmeel-octomap==1.10.0
 + cmeel-qhull==8.0.2.1
 + cmeel-tinyxml2==10.0.0
 + cmeel-urdfdom==4.0.1
 + cmeel-zlib==1.3.1
 + coal-library==3.0.1
 + eigenpy==3.10.3
 + numpy==2.2.5
 + pin==3.4.0
(robosuite) ck@ks-X1E-Gen-3:~/workspaces/robosuite$ python
Python 3.12.6 (main, Sep  9 2024, 22:11:19) [Clang 18.1.8 ] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pinocchio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/ros/humble/lib/python3.10/site-packages/pinocchio/__init__.py", line 19, in <module>
    from .pinocchio_pywrap_default import *
ModuleNotFoundError: No module named 'pinocchio.pinocchio_pywrap_default'
```
## Expected behavior

the import should be successful.

## Reproduction steps

```
pip install uv
uv venv --python 3.12.6
source .venv/bin/activate
uv pip install pin

### System

- OS: Ubuntu 22.04
- Pinocchio version: 3.4.0

nim65s, 2025-05-05:
I can't reproduce:
```
$ docker run --rm -it ghcr.io/astral-sh/uv:debian bash
root@661b9f217e61:/# uv venv --python 3.12.6
Using CPython 3.12.6
Creating virtual environment at: .venv
root@661b9f217e61:/# source .venv/bin/activate
(.venv) root@661b9f217e61:/# uv pip install pin
Resolved 13 packages in 295ms
Prepared 13 packages in 11.01s
Installed 13 packages in 167ms
 + cmeel==0.57.3
 + cmeel-assimp==5.4.3.1
 + cmeel-boost==1.87.0.1
 + cmeel-console-bridge==1.0.2.3
 + cmeel-octomap==1.10.0
 + cmeel-qhull==8.0.2.1
 + cmeel-tinyxml2==10.0.0
 + cmeel-urdfdom==4.0.1
 + cmeel-zlib==1.3.1
 + coal-library==3.0.1
 + eigenpy==3.10.3
 + numpy==2.2.5
 + pin==3.4.0
(.venv) root@661b9f217e61:/# python -c "import pinocchio; print(pinocchio.__version__)"
3.4.0
```

Something is wrong in your environment. You should probably clean your `PYTHONPATH` / `LD_LIBRARY_PATH`

kczttm, 2025-05-05:
@nim65s Thanks for the fast response. 

I have tried on a different 22.04 computer, and it didn't have this issue, so I'll just proceed with that computer for now. I'll reopen this if cleaning PYTHONPATH / LD_LIBRARY_PATH doesn't fix it in the future.

closed 2025-05-05 by kczttm

nim65s, 2025-05-05:
Oh, yes, I missed that, thanks @traversaro 
```
  File "/opt/ros/humble/lib/python3.10/site-packages/pinocchio/__init__.py", line 19, in <module>
[…]
uv pip install pin
```

Don't install 2 versions of pinocchio. Choose ROS **or** pip. They can not work together.