stack-of-tasks/pinocchio #2674: Casadi & ROS

nim65s, 2025-05-05:
> Hello @jorisv,
> I'm using the ROS 2 Humble version of Pinocchio and encountered an error when trying to import pinocchio.casadi.
> Does this mean that cpin is not yet supported in this version? 
> I would appreciate any clarification you could provide. Thank you! 
> (As a second option, I'm using C++ instead, just for your information.)
> 
> ```bash
>  $python -c "from pinocchio import casadi as cpin"
> Traceback (most recent call last):
>   File "<string>", line 1, in <module>
> ImportError: cannot import name 'casadi' from 'pinocchio' (/opt/ros/humble/lib/python3.10/site-packages/pinocchio/__init__.py)
> ``` 

 _Originally posted by @aky-u in [#2078](https://github.com/stack-of-tasks/pinocchio/issues/2078#issuecomment-2851364604)_

nim65s, 2025-05-05:
@aky-u : casadi is not packaged in ROS, therefore the `ros-*-pinocchio` packages do not have the casadi option. You can use `robotpkg-py\*-pinocchio` instead, the package will include casadi, and be compatible with the other packages of ros. Other solutions include compiling pinocchio yourself, or replacing your whole ROS setup one provided by conda or nix.

aky-u, 2025-05-05:
@nim65s: Thank you for your information!

closed 2025-05-05 by nim65s