stack-of-tasks/pinocchio #2676: Handling Gripper Width with Mimic Joints

HarisCavkic, 2025-05-06:
Hi Pinocchio team,

I’m currently working with a gripper that uses mimic joints to define finger motion, and I’m trying to determine the correct procedure to:
	1.	Compute the gripper width from the current (mimicked) joint value.
	2.	Compute the required joint value given a target gripper width.

Is this functionality treated as part of FK/IK in Pinocchio, or is there a different recommended approach for dealing with mimic joints in this context? I couldn’t find an example in the documentation or demos that covers this specific case.

Any guidance or pointers would be greatly appreciated!

Kind regards,
Haris

jorisv, 2025-05-06:
Hello @HarisCavkic,

Mimic joints are handled by pinocchio for FK et jacobian computation. So you should be able compute IK for a robot with mimic joints.

The following file show how to use mimic joints in Pyhon: https://github.com/stack-of-tasks/pinocchio/blob/devel/examples/mimic_dynamics.py
From urdf parser you have to use the following argument: `mimic=True`
If you create your model manually you can use [transformJointIntoMimic or buildMimicModel functions](https://github.com/stack-of-tasks/pinocchio/blob/devel/include/pinocchio/algorithm/model.hpp#L220)

There is one constraints: Mimicking joint should have an higher index than mimicked joint.

When you have a working model with mimic joint, you can use a standard [IK algorithm](https://github.com/stack-of-tasks/pinocchio/blob/devel/examples/inverse-kinematics.py)

j-matheron, 2026-09-10:
hello @HarisCavkic, 
if all is fine for you, you can close this issue