stack-of-tasks/pinocchio #2678: Issues with liburdfdom_model during C++ compilation and some questions regarding handling parallel robots

Wan020607, 2025-05-14:
## Bug description

I'm trying to install Pinocchio using the build from source method, but only as a sub-library within my C++ project for local usage like ocs2.
During the catkin build process, I encountered the following warning:
`/usr/bin/ld: warning: liburdfdom_model.so.4.0, needed by /home/wan/learning_for_pin/devel/lib/pkgconfig/../../lib/libpinocchio_parsers.so, may conflict with liburdfdom_model.so.1.0`
Later, when I ran the code, I got the following runtime error:
`error while loading shared libraries: liburdfdom_model.so.4.0: cannot open shared object file: No such file or directory
`Then I checked the dependencies of the executable using ldd:
```
ldd /home/wan/learning_for_pin/devel/lib/mass_calculation/main_node
	linux-vdso.so.1 (0x00007ffdc15ea000)
	libpinocchio_parsers.so.3.6.0 => /home/wan/learning_for_pin/devel/lib/libpinocchio_parsers.so.3.6.0 (0x00007f628d94d000)
	liburdfdom_model.so.1.0 => /lib/x86_64-linux-gnu/liburdfdom_model.so.1.0 (0x00007f628d8e6000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f628d704000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f628d5b5000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f628d59a000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f628d3a6000)
	libpinocchio_default.so.3.6.0 => /home/wan/learning_for_pin/devel/lib/libpinocchio_default.so.3.6.0 (0x00007f628cf71000)
	libboost_filesystem.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_filesystem.so.1.71.0 (0x00007f628cf53000)
	liburdfdom_model.so.4.0 => not found
	libconsole_bridge.so.0.4 => /lib/x86_64-linux-gnu/libconsole_bridge.so.0.4 (0x00007f628cf4d000)
	libtinyxml.so.2.6.2 => /lib/x86_64-linux-gnu/libtinyxml.so.2.6.2 (0x00007f628cf35000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f628da2c000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f628cf10000)
```
I also checked the dependencies of the **libpinocchio_parsers.so** library itself:
```
ldd /home/wan/learning_for_pin/devel/lib/libpinocchio_parsers.so | grep urdf
liburdfdom_model.so.4.0 => not found
```
So, does Pinocchio now require a **URDFDOM version >= 4.0**?
However, according to the documentation at https://stack-of-tasks.github.io/pinocchio/download.html, the required dependency is:

> URDFDOM for the reading of URDF models. Version >= 0.2.10

## Expected behavior

I would like to use the Pinocchio dynamics library to handle robot models with **parallel mechanisms** ( **mjcf or sdf** ), in order to obtain **their generalized mass matrix, nonlinear effects, and generalized gravity terms**.

## Reproduction steps

Steps to reproduce the behavior:
1. Clone Pinocchio from source into the src folder of my workspace, and create another package inside src to use and call Pinocchio locally.
2. Create the corresponding CMakeLists.txt and package.xml files for the new package, and compile the workspace using catkin build.

### Code

The code I use to call the Pinocchio dynamics library is as follows:

```
#include <pinocchio/multibody/model.hpp>
#include <pinocchio/multibody/data.hpp>
#include "pinocchio/parsers/mjcf.hpp"
#include "pinocchio/parsers/mjcf/mjcf-graph.hpp"
#include <pinocchio/algorithm/kinematics.hpp>
#include <iostream>

int main(int argc, char **argv)
{
    std::string mjcf_filename = "/home/wan/learning_for_pin/src/mass_calculation/model/urdf/hexapod_simgle_leg.xml";
    pinocchio::Model model;
    pinocchio::Data data(model);

    pinocchio::mjcf::buildModel(mjcf_filename, model, true);

    std::cout << "Model name: " << model.name << std::endl;

    Eigen::VectorXd q = pinocchio::randomConfiguration(model);

    pinocchio::forwardKinematics(model, data, q);

    for (pinocchio::JointIndex i = 0; i < model.njoints; ++i)
    {
        std::cout << model.names[i] << ": " << data.oMi[i].translation().transpose() << std::endl;
    }

    return 0;
}
```

## Additional context
Aside from that, I have another question. I also tried using the Pinocchio dynamics library in Python, with the same version as the one used in C++. The code is as follows:
```
import pinocchio as pin
from pinocchio.visualize import MeshcatVisualizer
import numpy as np

mjcf_model_path = "/home/wan/closed_loop_test/src/urdf/hexapod_simgle_leg.xml"
model = pin.buildModelFromMJCF(mjcf_model_path)

data = model.createData()

q = pin.neutral(model)
v = np.zeros(model.nv)

pin.computeAllTerms(model, data, q, v)
M = data.M

print("Generalized mass matrix:\n", M)
print("Gravity term (data.g):", data.g)
coriolis = data.nle - data.g
print("Coriolis / centrifugal term (computed):", coriolis)
```
The output is as follows:
```
Generalized mass matrix:
 [[ 5.99291300e+01 -2.92423955e+01  3.17520839e+00  3.01688884e+00
   5.75472197e-08  0.00000000e+00]
 [-2.92423955e+01  3.66080148e+01  1.26113353e+01 -1.18653401e+00
   4.75134287e-02  0.00000000e+00]
 [ 3.17520839e+00  1.26113353e+01  1.54690229e+01  1.52866594e+00
   0.00000000e+00  0.00000000e+00]
 [ 3.01688884e+00 -1.18653401e+00  1.52866594e+00  1.52866594e+00
   0.00000000e+00  0.00000000e+00]
 [ 5.75472197e-08  4.75134287e-02  0.00000000e+00  0.00000000e+00
   4.75134391e-02  0.00000000e+00]
 [ 0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00
   0.00000000e+00  1.32040000e+01]]
Gravity term (data.g): [ 5.87904765e+02 -2.86867900e+02  3.11487943e+01  2.95956795e+01
  5.64538225e-07  0.00000000e+00]
Coriolis / centrifugal term (computed): [ 0.00000000e+00  5.68434189e-14 -7.10542736e-15  0.00000000e+00
 -3.76806842e-17  0.00000000e+00]
```
This robot has only **2 generalized coordinates**, but due to the 2 parallel joints, it results in 6 degrees of freedom. The output generalized mass matrix still treats it as a robot with 6 generalized coordinates.
So, whether in Python or C++, when handling parallel robots, is it necessary like **example/simulation-closed-kinematic-chains.py** explicitly **apply constraints** after loading an **MJCF or SDF file that describes a closed-chain robot**? I noticed that Pinocchio seems to support identifying constraints such as **equality tags from MJCF** files.

I will appreciate it for your help!!!!

### System
- Ubuntu 20.04
- Ros: Noetic
- Pinocchio version: 3.6.0

jcarpent, 2025-05-14:
@Wan020607, Could you please open one issue per topic? Otherwise, it will be a nightmare for us.

jcarpent, 2025-05-14:
Related to the main issue, I would invite you to consider installing only one copy of Pinocchio and its dependencies on your system.
The current issue is indeed due to the fact that two URDFdom libraries are installed on your system and are accessible in your `LD_LIBRARY_PATH`.

Wan020607, 2025-05-15:
> Related to the main issue, I would invite you to consider installing only one copy of Pinocchio and its dependencies on your system. The current issue is indeed due to the fact that two URDFdom libraries are installed on your system and are accessible in your `LD_LIBRARY_PATH`.

Wan020607, 2025-05-15:
OK, I will open another issue to ask about dealing with closed chain joint using pinocchio.
And I finish the error of liburdfdom_model.so.4.0 by rebuilding my pinocchio, and clear the path to liburdfdom_model.so.4.0, make the Pinocchio library access only the path of liburdfdom_model.so.1.0 during compilation, and prevent it from accessing liburdfdom_model.so.4.0.