stack-of-tasks/pinocchio #2682: libbost question

kelo020304, 2025-05-23:
I am using Ubuntu20.04 conda env with python 3.10 and encounter the question：

    from .coal_pywrap import *  # noqa
ImportError: /lib/x86_64-linux-gnu/libboost_python38.so.1.71.0: undefined symbol: _Py_fopen

using pip install to install the Pinocchio, how to solve it

jcarpent, 2025-05-23:
Hard to say at first glance.
Could you try installing Pinocchio in a proper and clean environment?
```python
conda create -n env pin3
conda activate pin3
conda install -c conda-forge pinocchio
python -c "import pinocchio"
```
PS: you're still working on ubuntu 20.04 which is no more supported.

kelo020304, 2025-05-23:
Thank you for quick response
I just try a new clean environment
but I install Pinocchio with following way:
```
cd pyroboplan && pip install -e .
```
But encounter the same issue 
If I using conda install -c conda-forge Pinocchio it will take too long time to install

kelo020304, 2025-05-23:
Oh it not very same 
The error is:
```
from .pinocchio_pywrap_default import *
ImportError: /lib/x86_64-linux-gnu/libboost_python38.so.1.71.0: undefined symbol: _Py_fopen
```

jcarpent, 2025-05-23:
It seems you have env variables. Please unset them and everything will go well.
I will close the issue as it is not an error from our side, but from system configuration.

closed 2025-05-23 by jcarpent