stack-of-tasks/pinocchio #2683: Possible regression in `computeFrameJacobian` in Pinocchio 3.5.0

scastro-bdai, 2025-05-23:
## Bug description

@robodreamer and I have been running into a bug as our use of ROS 2 Humble latest version upgraded us from Pinocchio 3.4.0 to 3.5.0.

We were computing a specific frame Jacobian for our robot which came out to some nonzero matrix, but suddenly upon upgrade the matrix is zero.

Curiously enough, the Python API works totally fine in both versions, and we are not sure why this is the case (maybe the Python bindings still use an older version somehow? Or there is some other usage mistake?).

The suspicious change is this commit: https://github.com/stack-of-tasks/pinocchio/commit/39dd6a9f3f0aadc4973e0e8f48c84c64aed9a918#diff-6cdec46d90a62bca95369951a2d224a178f210bb8483bee8ae6991db1cbce3b0R212 -- as it lines up with changes to the `frames.hxx` file between 3.4.0 and 3.5.0. And it does seem to be a significant behavioral change in the Jacobian calculation implementation.

## Expected behavior

Frame Jacobian for our model should not change.

## Reproduction steps

Steps to reproduce the behavior:
1. Get the files from https://github.com/scastro-bdai/pinocchio-jacobian-repro
2. Run the Python file. You should see this when computing the Jacobian of the `right_fr3_link2` frame:

```
q:
[0 -0.35 0 -1.87 0 1.45 0.785 0 0 0 -0.35 0 -1.87 0 1.45 0.785 0 0]
Geometric Jacobian:
[[0 0 0 0 0 0 0 0 0 0 -0 0 0 0 0 0 0 0]
 [0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0]
 [0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0]
 [0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0]
 [0 0 0 0 0 0 0 0 0 0 1 0 0 0 0 0 0 0]
 [0 0 0 0 0 0 0 0 0 1 -4.441e-16 0 0 0 0 0 0 0]]
```

3. When using the same in C++, I get all zeros in 3.5.0 but the same value in 3.4.0. Sorry I ran out of time to provide a C++ repro case that strips out from our internal code... but hopefully for maintainers testing this, you can do this with just the provided URDF and your own simple C++ file / build system of choice?

```
Frame name: right_fr3_link2
Computing full Jacobian with state:
    0
-0.35
    0
-1.87
    0
 1.45
0.785
    0
    0
    0
-0.35
    0
-1.87
    0
 1.45
0.785
    0
    0
Frame Jacobian:
 0  0  0  0  0  0  0  0  0  0 -0  0  0  0  0  0  0  0
 0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
 0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
```

### System

- OS: Ubuntu 22.04
- Pinocchio version: 3.5.0

jcarpent, 2025-05-24:
Dear @scastro-bdai,

Thanks for raising this issue. 

> Is this an intentional change, or should resizing to `6 x nv` be reinstated? Thanks!

I would say that any Jacobians related to frames or joints in Pinocchio are 6xnv matrix objects, as they are related to both linear and angular motions of the frame (variation of spatial velocities, accelerations, etc.).
For any version of Pinocchio, you should provide a `6 x nv` matrix and not a `3 x nv` matrix filled with Zeros.

On our side, we have overlooked checking the rows of the matrices provided by the user. This is our mistake ;).
I have fixed this behavior in #2684.

I hope my explanations make sense to you.
Very best,
Justin

jcarpent, 2025-05-26:
Solved via #2684