stack-of-tasks/pinocchio #2685: Write a OpSpace ros2_controllers with pinocchio

ItsTomSav, 2025-05-26:
Hi all,
I apologize if this question might be off-topic and more related to ros2_control.
I couldn't find any useful examples for implementing a NEW ros2 controller (in the operational space) that also uses pinocchio and not only ros libraries, therefore usable from the controller manager (I'm following ["writing_new_controller").](https://control.ros.org/humble/doc/ros2_controllers/doc/writing_new_controller.html) 
I only found this ["pinocchio_example"](https://github.com/sea-bass/pinocchio_ros_cpp_example/blob/main/pinocchio_ros_example/src/pinocchio_example.cpp) (a cpp file) but there is no example or recommendation for the .hpp file.
If anyone has examples or even suggestions, I would be very grateful.

Regards,
Tom.

sea-bass, 2025-05-26:
How is an example with just a .cpp file different from one with an .hpp and a .cpp? You just need to split up the declarations and definitions across files and include the right folders in your CMakeLists.txt.

Regardless, this might be more useful as an example: https://github.com/justagist/kinematics_interface_pinocchio

ItsTomSav, 2025-05-27:
> How is an example with just a .cpp file different from one with an .hpp and a .cpp? You just need to split up the declarations and definitions across files and include the right folders in your CMakeLists.txt.
> 
> As far are what Pinocchio headers to include and how to link them, should be identical.
> 
> Regardless, this might be more useful as an example for `ros2_control`: https://github.com/justagist/kinematics_interface_pinocchio

Thanks for the clarification, I don't know why but I thought there might be some differences compared to other examples without Pinocchio.

closed 2025-05-27 by ItsTomSav