stack-of-tasks/pinocchio #2690: when i catkin build project ocs2 , need pinocchio with hpp-fcl collision support, but Error: pinocchio/collision/distance.hpp:82:8: error: ‘DistanceResult’ in namespace ‘pinocchio::fcl’ does not name a type    82 |   fcl::DistanceResult & computeDistance(

zcwang1027, 2025-05-29:
## Bug description

when i catkin build project ocs2 (ocs2_legged_robot_ros ocs2_self_collision_visualization), 
need pinocchio with hpp-fcl collision support, 
but occor Error: pinocchio/collision/distance.hpp:82:8: **error: ‘DistanceResult’** in namespace ‘pinocchio::fcl’ does not name a type    82 |   fcl::DistanceResult & computeDistance(

## Reproduction steps

Steps to reproduce the behavior:
1. build hpp-fcl (v2.4.5)
https://github.com/coal-library/coal.git
cmake make make install OK
2. build pinocchio (v3.7.0) 
https://github.com/stack-of-tasks/pinocchio.git
Enable collision in cmakelist: option(BUILD_WITH_COLLISION_SUPPORT
       "Build the library with the collision support (required HPP-FCL)" ON)
cmake make make install OK
3. catkin build ocs2_legged_robot_ros ocs2_self_collision_visualization (v12.0)
https://github.com/leggedrobotics/ocs2.git
catkin config -DCMAKE_BUILD_TYPE=RelWithDebInfo
catkin build ocs2_legged_robot_ros ocs2_self_collision_visualization
4. Error
Errors     << ocs2_self_collision:make /home/zcwang/rosbot_ws/logs/ocs2_self_collision/build.make.000.log                                                        
In file included from /home/zcwang/rosbot_ws/src/ocs2/ocs2_pinocchio/ocs2_self_collision/src/PinocchioGeometryInterface.cpp:36:
**/usr/local/include/pinocchio/collision/distance.hpp:82:8: error: ‘DistanceResult’ in namespace ‘pinocchio::fcl’ does not name a type
   82 |   fcl::DistanceResult & computeDistance(
      |        ^~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.**
5. I check file pinocchio/collision/distance.hpp
and find its' namespace is fcl, not hpp::fcl used in hpp-fcl file
other file also use namespace fcl, not hpp::fcl.
code in file distance.hpp:
```
  fcl::DistanceResult & computeDistance(
    const GeometryModel & geom_model, GeometryData & geom_data, const PairIndex pair_id);
```
6. Is there something wrong when i build ocs2?

### Code

If this is a runtime issue, provide some minimal code to reproduce it:

```
catkin build ocs2_legged_robot_ros ocs2_self_collision_visualization[build - 01:33.4] [16/21 complete] [2/28 jobs] [0 queued] [ocs2_centroidal_model:make (87%) - _________________________________________________________________________________________________________________________________________________________________
Errors     << ocs2_self_collision:make /home/zcwang/rosbot_ws/logs/ocs2_self_collision/build.make.000.log                                                        
In file included from /home/zcwang/rosbot_ws/src/ocs2/ocs2_pinocchio/ocs2_self_collision/src/PinocchioGeometryInterface.cpp:36:
/usr/local/include/pinocchio/collision/distance.hpp:82:8: error: ‘DistanceResult’ in namespace ‘pinocchio::fcl’ does not name a type
   82 |   fcl::DistanceResult & computeDistance(
      |        ^~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
make[2]: *** [CMakeFiles/ocs2_self_collision.dir/build.make:76: CMakeFiles/ocs2_self_collision.dir/src/PinocchioGeometryInterface.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:269: CMakeFiles/ocs2_self_collision.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
cd /home/zcwang/rosbot_ws/build/ocs2_self_collision; catkin build --get-env ocs2_self_collision | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

.................................................................................................................................................................
Failed     << ocs2_self_collision:make                         [ Exited with code 2 ]                                                                            
Failed    <<< ocs2_self_collision                              [ 18.4 seconds ]                                                                                  
Abandoned <<< ocs2_self_collision_visualization                [ Unrelated job failed ]                                                                          
Abandoned <<< ocs2_legged_robot                                [ Unrelated job failed ]                                                                          
Abandoned <<< ocs2_legged_robot_ros                            [ Unrelated job failed ]                                                                          
_________________________________________________________________________________________________________________________________________________________________
Warnings   << ocs2_centroidal_model:make /home/zcwang/rosbot_ws/logs/ocs2_centroidal_model/build.make.000.log                                                    
/home/zcwang/rosbot_ws/src/ocs2/ocs2_pinocchio/ocs2_centroidal_model/src/CentroidalModelRbdConversions.cpp: In member function ‘ocs2::vector_t ocs2::CentroidalModelRbdConversions::computeRbdTorqueFromCentroidalModelPD(const ocs2::vector_t&, const ocs2::vector_t&, const ocs2::vector_t&, const ocs2::vector_t&, const ocs2::vector_t&, const ocs2::vector_t&)’:
/home/zcwang/rosbot_ws/src/ocs2/ocs2_pinocchio/ocs2_centroidal_model/src/CentroidalModelRbdConversions.cpp:192:54: warning: ‘pinocchio::FrameTpl<double, 0>::parent’ is deprecated [-Wdeprecated-declarations]
  192 |     const auto jointIndex = model.frames[frameIndex].parent;
      |                                                      ^~~~~~
In file included from /usr/local/include/pinocchio/multibody/model.hpp:16,
                 from /usr/local/include/pinocchio/algorithm/centroidal.hpp:8,
[... 24 lines cut ...]
      |                                                      ^~~~~~
/usr/local/include/pinocchio/multibody/frame.hpp:231:39: note: declared here
  231 |     PINOCCHIO_DEPRECATED JointIndex & parent;
      |                                       ^~~~~~
/home/zcwang/rosbot_ws/src/ocs2/ocs2_pinocchio/ocs2_centroidal_model/src/CentroidalModelRbdConversions.cpp:201:54: warning: ‘pinocchio::FrameTpl<double, 0>::parent’ is deprecated [-Wdeprecated-declarations]
  201 |     const auto jointIndex = model.frames[frameIndex].parent;
      |                                                      ^~~~~~
/usr/local/include/pinocchio/multibody/frame.hpp:231:39: note: declared here
  231 |     PINOCCHIO_DEPRECATED JointIndex & parent;
      |                                       ^~~~~~
At global scope:
cc1plus: note: unrecognized command-line option ‘-Wno-invalid-partial-specialization’ may have been intended to silence earlier diagnostics
cd /home/zcwang/rosbot_ws/build/ocs2_centroidal_model; catkin build --get-env ocs2_centroidal_model | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

.................................................................................................................................................................
Finished  <<< ocs2_centroidal_model                            [ 47.0 seconds ]                                                                                  
[build] Summary: 17 of 21 packages succeeded.                                                                                                                    
[build]   Ignored:   38 packages were skipped or are skiplisted.                                                                                                 
[build]   Warnings:  14 packages succeeded with warnings.                                                                                                        
[build]   Abandoned: 3 packages were abandoned.                                                                                                                  
[build]   Failed:    1 packages failed.                                                                                                                          
[build] Runtime: 2 minutes and 4.1 seconds total.                                                                                                                
[build] Note: Workspace packages have changed, please re-source setup files to use them.

```

ManifoldFR, 2025-05-29:
Hello mate,

I don't think you're supposed to use that version of hpp-fcl (2.4) with this version of Pinocchio. Try upgrading to coal (the new name of hpp-fcl), the latest version is 3.0.1.

If that doesn't work you might have to wait for someone to take another look, it's Ascension Day weekend in our country.

Cheers!

zcwang1027, 2025-05-31:
> Hello mate,
>
> I don't think you're supposed to use that version of hpp-fcl (2.4) with this version of Pinocchio. Try upgrading to coal (the new name of hpp-fcl), the latest version is 3.0.1.
>
> If that [...]

Sincerely thanks, OCS2 depends on pinocchio version 2, directly using pinocchio version 3 and enabling support for version 2 seems to have some bugs. After I switched to pinocchio version 2, ocs2 started working.