stack-of-tasks/pinocchio #2702: Analytical ∂v/∂q from getJointVelocityDerivatives disagrees with CppAD

Zionshang, 2025-06-15:
## Bug description

The derivatives of joint velocity (in `LOCAL_WORLD_ALIGNED` frame) w.r.t. `q` from `getJointVelocityDerivatives` do not match the autodiff results from CppAD. 

However, the derivative w.r.t. velocity `v` does match.

**I have already performed integration during CppAD differentiation to ensure that the result with respect to `q` satisfies the manifold constraints.**

## Reproduction steps

1. Load a URDF with `JointModelFreeFlyer` (e.g., go2.urdf)
2. Compare results of:
  - `getJointVelocityDerivatives(...)`
  - CppAD on `getVelocity(...)` after `integrate(q, dq)`

### Code
```cpp
#include <pinocchio/parsers/urdf.hpp>
#include <pinocchio/algorithm/kinematics.hpp>
#include <pinocchio/algorithm/kinematics-derivatives.hpp>
#include <pinocchio/algorithm/joint-configuration.hpp>
#include <pinocchio/autodiff/cppad.hpp>

using pinocchio::context::Matrix6xs;
int main(int argc, char const *argv[])
{
    ////////////////// Load Model //////////////////
    std::string urdf_filename = EXAMPLE_ROBOT_DATA_MODEL_DIR "/go2_description/urdf/go2.urdf";
    pinocchio::Model model;
    pinocchio::urdf::buildModel(urdf_filename, pinocchio::JointModelFreeFlyer(), model);
    pinocchio::Data data(model);
    int nq = model.nq;
    int nv = model.nv;
    model.lowerPositionLimit.head<3>().fill(0.);
    model.upperPositionLimit.head<3>().fill(0.5);

    ////////////////// Random configuration and velocity //////////////////
    pinocchio::JointIndex joint_id = 10;
    std::srand(static_cast<unsigned int>(std::time(nullptr)));
    Eigen::VectorXd q = pinocchio::randomConfiguration(model);
    Eigen::VectorXd v = Eigen::VectorXd::Random(nv);

    ////////////////// Analytical derivatives //////////////////
    Matrix6xs v_partial_dq = Matrix6xs::Zero(6, nv);
    Matrix6xs v_partial_dv = Matrix6xs::Zero(6, nv);
    pinocchio::computeForwardKinematicsDerivatives(model, data, q, v, Eigen::VectorXd::Zero(nv));
    pinocchio::getJointVelocityDerivatives(model, data, joint_id, pinocchio::LOCAL_WORLD_ALIGNED, v_partial_dq, v_partial_dv);

    ////////////////// CPPAD derivatives //////////////////
    using CppAD::AD;
    using ADVectorX = Eigen::VectorX<AD<double>>;

    pinocchio::ModelTpl<AD<double>> ad_model = model.cast<AD<double>>();
    pinocchio::DataTpl<AD<double>> ad_data(ad_model);

    ADVectorX ad_X(nq + nv + nv); // q, v, dq
    ad_X.setZero();
    CppAD::Independent(ad_X);
    ADVectorX ad_Y(6);

    ADVectorX ad_q = ad_X.head(nq);
    ADVectorX ad_v = ad_X.segment(nq, nv);
    ADVectorX ad_dq = ad_X.tail(nv);
    ADVectorX ad_q_plus = pinocchio::integrate(ad_model, ad_q, ad_dq);

    pinocchio::forwardKinematics(ad_model, ad_data, ad_q_plus, ad_v);
    auto joint_motion = pinocchio::getVelocity(ad_model, ad_data, joint_id, pinocchio::LOCAL_WORLD_ALIGNED);
    ad_Y.head(3) = joint_motion.linear();
    ad_Y.tail(3) = joint_motion.angular();
    CppAD::ADFun<double> ad_joint_vel(ad_X, ad_Y);
[... 33 lines cut ...]
           0            0            0    0.0205968     -1.44842    0.0325255            0            0            0            0            0            0     0.330175 -2.08167e-16 -3.05311e-16            0            0            0
cppad derivatives:
           0            0            0     0.344856      1.04991     0.830332            0            0            0            0            0            0  -0.00158098     -0.09355 -1.66533e-16            0            0            0
           0            0            0       -1.027     0.384139    -0.126749            0            0            0            0            0            0    -0.107076     0.153736  3.51282e-17            0            0            0
           0            0            0    -0.757111     0.355097   -0.0453908            0            0            0            0            0            0    -0.238187   -0.0632826            0            0            0            0
           0            0            0   0.00587014    -0.065837    -0.527009            0            0            0            0            0            0   0.00070275  2.22045e-16  5.55112e-17            0            0            0
           0            0            0   -0.0127376   0.00898251      1.35048            0            0            0            0            0            0     0.044904 -1.11022e-16 -1.38778e-16            0            0            0
           0            0            0    0.0205968     -1.44842    0.0325255            0            0            0            0            0            0     0.330175 -1.45717e-16 -6.41848e-17            0            0            0
==============Joint velocity partial derivative w.r.t. velocity==============
pinocchio derivatives:
   0.935188    -0.34999   0.0541394   0.0638927    0.150382     -0.1315           0           0           0           0           0           0   0.0613752    0.161278           0           0           0           0
   0.350651    0.936502 -0.00292169   -0.147351   0.0547892    -0.12274           0           0           0           0           0           0   -0.147215   0.0437756           0           0           0           0
 -0.0496791   0.0217164    0.998529    0.162702   0.0608834  0.00677067           0           0           0           0           0           0     0.11627   -0.132069           0           0           0           0
          0           0           0    0.935188    -0.34999   0.0541394           0           0           0           0           0           0    0.935188   -0.354146   -0.354146           0           0           0
          0           0           0    0.350651    0.936502 -0.00292169           0           0           0           0           0           0    0.350651     0.92676     0.92676           0           0           0
          0           0           0  -0.0496791   0.0217164    0.998529           0           0           0           0           0           0  -0.0496791   -0.125286   -0.125286           0           0           0
cppad derivatives:
   0.935188    -0.34999   0.0541394   0.0638927    0.150382     -0.1315           0           0           0           0           0           0   0.0613752    0.161278           0           0           0           0
   0.350651    0.936502 -0.00292169   -0.147351   0.0547892    -0.12274           0           0           0           0           0           0   -0.147215   0.0437756           0           0           0           0
 -0.0496791   0.0217164    0.998529    0.162702   0.0608834  0.00677067           0           0           0           0           0           0     0.11627   -0.132069           0           0           0           0
          0           0           0    0.935188    -0.34999   0.0541394           0           0           0           0           0           0    0.935188   -0.354146   -0.354146           0           0           0
          0           0           0    0.350651    0.936502 -0.00292169           0           0           0           0           0           0    0.350651     0.92676     0.92676           0           0           0
          0           0           0  -0.0496791   0.0217164    0.998529           0           0           0           0           0           0  -0.0496791   -0.125286   -0.125286           0           0           0
```
Observed: `v_partial_dq` ≠ `djoint_vel_dq`, but `v_partial_dv` = `djoint_vel_dv`.

Zionshang, 2025-06-15:
A strange thing happened — when I switched the reference frame to `WORLD` or `LOCAL`, the results are correct!

Zionshang, 2025-06-15:
I believe the CppAD result is more reliable, since this is a floating-base robot (a quadruped). For joint velocity in the `LOCAL_WORLD_ALIGNED` frame, the floating base translation shouldn't affect it — so the derivative w.r.t. the floating base's translation should be 0.

ManifoldFR, 2025-06-15:
> A strange thing happened — when I switched the reference frame to `WORLD` or `LOCAL`, the results are correct!

Not so strange, the way joint velocities and corresponding ops (integrate, difference) involved in the expression you want to differentiate correspond to the `LOCAL` reference frame. If you pass one of the other reference frames to the derivatives function, there are some additional corrections that are done, which is something the current autodiff systems cannot really handle out of the box.

Zionshang, 2025-06-15:
Thank you very much for your explanation! It’s indeed a bit hard to grasp, because intuitively I always felt that the floating base translation doesn't affect the LOCAL_WORLD_ALIGNED velocity of other joints.

closed 2025-06-15 by Zionshang

jcarpent, 2025-06-20:
The derivatives for LOCAL_WORLD_ALIGNED convention have been fixed in Pinocchio 4, which should be released this summer.
Thanks @Zionshang for reporting this issue (which was known for a while, but I forgot it ;))