stack-of-tasks/pinocchio #2712: Release Pinocchio 3.5.0+ to ROS 2 Kilted?

sea-bass, 2025-06-22:
Hello maintainers!

I had a quick question. I'm working on a project that uses `pinocchio::integrate()`, which I guess moved headers between 3.4.0 and 3.5.0.

Right now, ROS 2 Humble, Jazzy, and Rolling have Pinocchio 3.5.0 but Kilted has 3.4.0.

Do you all release Pinocchio to non-LTS versions like ROS 2 Kilted? If so, would you be willing to release to that version too? Otherwise, I can do some header magic if needed.

Thanks!

sea-bass, 2025-06-22:
As usual, writing down the issue immediately makes me realize what the problem was and fix it 🤦🏻 .

I guess the change is that something in 3.5.0+ transitively included `<pinocchio/algorithm/joint-configuration.hpp>` for me under the hood, so explicitly including the header resolves my issue for cross-version compatibility.

But either way, might be useful to have the Kilted distro track the others, for the year that is it supported.

nim65s, 2025-06-22:
I just did not notice kilted was out. I guess it has 3.4.0 because it did branch off before 3.5.0 landed.
I'll update this, thanks for the heads-up

nim65s, 2025-06-23:
https://github.com/ros/rosdistro/pull/46589
Not sure why bloom decided that would be 3.6.0 and not 3.7.0 though :/

closed 2025-07-16 by sea-bass