stack-of-tasks/pinocchio #2742: [Bug]: ROS Rolling likely needs a new release of Pinocchio to be built

sea-bass, 2025-08-07:
### 🐞 Bug

I'm running into some issues with unit tests that use Pinocchio with ROS 2 rolling. The error message is:

```
error while loading shared libraries: liburdfdom_sensor.so.5.0: cannot open shared object file: No such file or directory
```

I suspect this has to do with the fact that the `urdfdom_headers` repo was bumped to a major 2.0.0 version in Rolling. See for example:

* https://repo.ros2.org/status_page/ros_kilted_default.html?q=urdfdom_headers
* https://repo.ros2.org/status_page/ros_rolling_default.html?q=urdfdom_headers

I can't quite figure out what broke ABI, but I hope this lead is enough to chase something? Thanks!

### Relevant log output / Error message

```shell

```

### Checklist

- [x] I have checked that all my packages are installed from the same package manager (conda-forge, Nix, PyPI, robotpkg, ROS, ...)
- [x] I have checked that there is no similar [issue](https://github.com/stack-of-tasks/pinocchio/issues)/[discussion](https://github.com/stack-of-tasks/pinocchio/discussions) in the repo
- [x] I have provided a minimal and working example to reproduce the bug
- [x] I have used [Markdown code blocks](https://help.github.com/en/articles/creating-and-highlighting-code-blocks) for both code and stack traces/compiler errors.

sea-bass, 2025-08-07:
It could also be that for some reason the `BUILD_WITH_URDF_SUPPORT` was disabled on rolling?

https://github.com/stack-of-tasks/pinocchio/blob/8d2080fb5224a00ca1704da570895bf8455bf600/CMakeLists.txt#L227

nim65s, 2025-08-07:
Please don't skip the reprodution steps :)

``Dockerfile
FROM ros:rolling

RUN apt update -qy \
 && apt upgrade -qy \
 && apt install -qy ros-rolling-pinocchio
```
```
$ docker build -t tmp . && docker run --rm -it tmp
root@2b122a0fef40:/# ldd /opt/ros/rolling/lib/x86_64-linux-gnu/libpinocchio*.so | grep urdf
	liburdfdom_sensor.so.5.0 => /opt/ros/rolling/lib/x86_64-linux-gnu/liburdfdom_sensor.so.5.0 (0x00007f9079755000)
	liburdfdom_model.so.5.0 => /opt/ros/rolling/lib/x86_64-linux-gnu/liburdfdom_model.so.5.0 (0x00007f9079739000)
	liburdfdom_world.so.5.0 => /opt/ros/rolling/lib/x86_64-linux-gnu/liburdfdom_world.so.5.0 (0x00007f907971b000)
```

The error message says the file liburdfdom_sensor.so.5.0 is not found on your system. Do you have `ros-rolling-urdfdom` v5 ?

closed 2025-08-07 by sea-bass