stack-of-tasks/pinocchio #2746: [Bug]: armature not carried over when using buildReducedModel

mikael-jorda, 2025-08-22:
### 🐞 Bug

It looks like when I use the buildReducedModel function, the new model comes with armatures equal to 0, no matter the armature of the initial model.

### Reproduction steps

```python
import pinocchio as pin
import numpy as np

model = pin.buildModelFromMJCF("double_pendulum.xml")

print("initial model (j1 and j2)")
for i in range(model.nv):
    print(f"joint name: {model.names[i+1]}, armature: {model.armature[i]}")
    

reduced_model = pin.buildReducedModel(model, [2], np.zeros(2))

print("\nreduced model (j1 only)")
for i in range(reduced_model.nv):
    print(f"joint name: {reduced_model.names[i+1]}, armature: {reduced_model.armature[i]}")
```

below is the model I used
```
<?xml version="1.0" ?>
<mujoco model="double_pendulum">
    <compiler angle="radian" />

    <option timestep="0.01" gravity="0 0 -9.81" />

    <default>
        <joint armature="0.1" damping="0.1" limited="false" />
        <geom contype="0" conaffinity="0" friction="0 0 0" />
    </default>

    <worldbody>
        <body name="link1" pos="0 0 0">
            <joint name="j1" type="hinge" axis="1 0 0" pos="0 0 0" />
            <inertial pos="0 0 -1" mass="1" diaginertia="0.1 0.2 0.3" />
            <site name="site1" />
            <body name="fixed_body_1">
                <site name="fixed_site_1" pos="0 0 0" />
            </body>
            <body name="link2" pos="0 0 -1">
                <joint name="j2" type="hinge" axis="1 0 0" pos="0 0 0" />
                <inertial pos="0 0 -1" mass="1" diaginertia="0.1 0.2 0.3" />
                <site name="site2" />
                <body name="eef" pos="0 0 -1">
                    <site name="site3" />
                </body>
            </body>
        </body>
    </worldbody>

    <actuator>
        <motor name="j1_motor" joint="j1" gear="1" ctrllimited="false" />
        <motor name="j2_motor" joint="j2" gear="1" ctrllimited="false" />
    </actuator>
</mujoco>
```

### Relevant log output / Error message

```shell
initial model (j1 and j2)
joint name: j1, armature: 0.1
joint name: j2, armature: 0.1

reduced model (j1 only)
joint name: j1, armature: 0.0
```

### Checklist

- [x] I have checked that all my packages are installed from the same package manager (conda-forge, Nix, PyPI, robotpkg, ROS, ...)
- [x] I have checked that there is no similar [issue](https://github.com/stack-of-tasks/pinocchio/issues)/[discussion](https://github.com/stack-of-tasks/pinocchio/discussions) in the repo
- [x] I have provided a minimal and working example to reproduce the bug
- [x] I have used [Markdown code blocks](https://help.github.com/en/articles/creating-and-highlighting-code-blocks) for both code and stack traces/compiler errors.

jcarpent, 2025-08-22:
Thanks a lot @mikael-jorda for raising this issue. It is indeed a bug and I will provide a fix today.

jcarpent, 2025-08-22:
The fix is here: https://github.com/stack-of-tasks/pinocchio/pull/2749

jorisv, 2025-08-26:
Fixed in #2749